NVIDIA-AI-IOT/tf_trt_models #25: Installation error: command 'aarch64-linux-gnu-gcc' failed with exit status 1

AtomicJosh, 2018-10-24:
On the TX2 running L4T 28.2, during the installation step:
```
pip3 install tensorflow-1.11.0-cp35-cp35m-linux_aarch64.whl --user
```
The following error was encountered:
```
In file included from /tmp/pip-install-s479x1s5/h5py/h5py/defs.c:654:0:
/tmp/pip-install-s479x1s5/h5py/h5py/api_compat.h:27:18: fatal error: hdf5.h: No such file or directory
compilation terminated.
error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
```
All of the libraries mentioned [here](https://stackoverflow.com/questions/26053982/setup-script-exited-with-error-command-x86-64-linux-gnu-gcc-failed-with-exit) were installed and the error persisted. The full output for that step is attached to this post.

[logTFerror.txt](https://github.com/NVIDIA-Jetson/tf_trt_models/files/2512503/logTFerror.txt)

88madri, 2018-10-26:
Same problem here.

RogerAylagas, 2019-04-10:
That worked for me!
I got the same issue.
Then I runned -> sudo apt-get install libhdf5-dev
and again runned -> pip3 install tensorflow-1.11.0-cp35-cp35m-linux_aarch64.whl --user

Installation worked successfully :)