NVIDIA-AI-IOT/tf_trt_models #44: Import Error: numpy.core.multiarray failed to import  when running camera_tf_trt.py in real time and in my own dataset

ruksin, 2019-03-18:
Hi,
First of all thank you for the tutorial. When I'm using camera_tf_trt.py in image huskies I got the detection properly. But can't do in the real time . My training dataset is different , not hands but I followed the steps, and again got the same error , could you please help me.

>                           --build
INFO:__main__:called with args: Namespace(conf_th=0.3, do_build=True, do_tensorboard=False, filename='data/image9.jpg', image_height=720, image_width=1280, labelmap_file='data/object-detection_1.pbtxt', model='ssd_mobilenet_v1_egohands', num_classes=2, rtsp_latency=200, rtsp_uri=None, use_file=False, use_image=True, use_rtsp=False, use_usb=False, video_dev=1)
INFO:__main__:reading label map
INFO:__main__:building TRT graph and saving to pb: ./data/ssd_mobilenet_v1_egohands_trt.pb
ImportError: No module named 'numpy.core._multiarray_umath'
Traceback (most recent call last):
  File "camera_tf_trt.py", line 244, in <module>
    main()
  File "camera_tf_trt.py", line 203, in main
    build_trt_pb(args.model, pb_path)
  File "/home/nvidia/project/tf_trt_models/utils/od_utils.py", line 43, in build_trt_pb
    from tf_trt_models.detection import download_detection_model
  File "/home/nvidia/project/tf_trt_models/tf_trt_models/detection.py", line 9, in <module>
    from object_detection import exporter
  File "/home/nvidia/.local/lib/python3.5/site-packages/object_detection-0.1-py3.5.egg/object_detection/exporter.py", line 28, in <module>
    from object_detection.builders import model_builder
  File "/home/nvidia/.local/lib/python3.5/site-packages/object_detection-0.1-py3.5.egg/object_detection/builders/model_builder.py", line 29, in <module>
    from object_detection.meta_architectures import ssd_meta_arch
  File "/home/nvidia/.local/lib/python3.5/site-packages/object_detection-0.1-py3.5.egg/object_detection/meta_architectures/ssd_meta_arch.py", line 32, in <module>
    from object_detection.utils import visualization_utils
  File "/home/nvidia/.local/lib/python3.5/site-packages/object_detection-0.1-py3.5.egg/object_detection/utils/visualization_utils.py", line 26, in <module>
    import matplotlib.pyplot as plt  # pylint: disable=g-import-not-at-top
  File "/home/nvidia/.local/lib/python3.5/site-packages/matplotlib-3.0.3-py3.5-linux-aarch64.egg/matplotlib/pyplot.py", line 32, in <module>
    import matplotlib.colorbar
  File "/home/nvidia/.local/lib/python3.5/site-packages/matplotlib-3.0.3-py3.5-linux-aarch64.egg/matplotlib/colorbar.py", line 28, in <module>
    import matplotlib.artist as martist
  File "/home/nvidia/.local/lib/python3.5/site-packages/matplotlib-3.0.3-py3.5-linux-aarch64.egg/matplotlib/artist.py", line 11, in <module>
    from .path import Path
  File "/home/nvidia/.local/lib/python3.5/site-packages/matplotlib-3.0.3-py3.5-linux-aarch64.egg/matplotlib/path.py", line 17, in <module>
    from . import _path, rcParams
ImportError: numpy.core.multiarray failed to import

I tried to update numpy. but still shows the same error. Could anyone please help me.
Thanking you in advance

MuhammadAsadJaved, 2019-10-22:
@ruksin  Can you tell me where is camera_tf_trt.py file? how we can run on our own models?

ruksin, 2019-10-22:
I don't remember much, what I have done to solve these errors. I followed a blog https://jkjung-avt.github.io
Follow these. Its a great help to train on custom dataset.

MuhammadAsadJaved, 2019-10-22:
Thank you very much. Before I was using another repo that's why I was confused.