NVIDIA-AI-IOT/tf_trt_models #48: Import error: cannot import 'pipeline_pb2'

RogerAylagas, 2019-04-24:
Hi,
I have followed the setup instructions and after finishing I have copied your jupyter notebook example code to a python script.
While trying to exectue it the following error occurs:

Traceback (most recent call last):
  File "run.py", line 14, in <module>
    from tf_trt_models.detection import download_detection_model, build_detection_graph
  File "/home/nvidia/.local/lib/python3.5/site-packages/tf_trt_models-0.0-py3.5.egg/tf_trt_models/detection.py", line 1, in <module>
ImportError: cannot import name 'pipeline_pb2'

Can anyone help me with this?
Thank you

RogerAylagas, 2019-04-24:
I noticed that the folder /home/nvidia/.local/lib/python3.5/site-packages/tf_trt_models-0.0-py3.5.egg is not there. Why?

kongkongthu, 2019-09-25:
I met the same question like yours, have you solved it? If so, can you show how you solved it.

jkjung-avt, 2019-09-25:
Did you run 'install.sh'?  I think the problem is caused by not compiling the '.proto' files with 'protoc'.  More specifically, that should be done by this line of command in 'install.sh':

https://github.com/NVIDIA-AI-IOT/tf_trt_models/blob/master/install.sh#L26

kongkongthu, 2019-09-26:
thank you very much, I forgot the install step.