NVIDIA-AI-IOT/tf_trt_models #55: Very slow 1.8s/image on Faster RCNN and wrongly perceived 1 image as 153216 images !!!

thusinh1969, 2019-07-10:
1) Ubuntu 16.04
2) Docker tensorflow/tensorflow 1.13.1
3) Docker NVIDIA:latest-gpu
4) Tensorflow object detection Faster RCNN Resnet101 successfully built w/ 2 classes only
5) Model are converted into FP32 (also tried FP16 with the same below issue)
with graph.as_default():
    with tf.Session() as sess:
        trt_graph = trt.create_inference_graph(
        input_graph_def=gdef,
        outputs=outputs,
        max_batch_size=300,
        max_workspace_size_bytes=4000000000,
        is_dynamic_op=True,
        #precision_mode='FP16') 
        precision_mode='FP32')
        #precision_mode='INT8')
        output_node=tf.import_graph_def(trt_graph, return_elements=outputs)
        #sess.run(output_node)
        tf.saved_model.simple_save(sess,
        rt_output_file_name_32,
        inputs={'input_image': graph.get_tensor_by_name('{}:0'.format(node.name))
        for node in graph.as_graph_def().node if node.op=='Placeholder'},
        outputs={t:graph.get_tensor_by_name('import/'+t) for t in outputs}
       )

Consistenly received ERROR/WARNING:

**2019-07-10 06:52:26.523782: W external/org_tensorflow/tensorflow/contrib/tensorrt/kernels/trt_engine_op.cc:264] Engine buffer is full. buffer limit=1, current entries=300, requested batch=153216
2019-07-10 06:52:26.523827: W external/org_tensorflow/tensorflow/contrib/tensorrt/kernels/trt_engine_op.cc:281] Failed to get engine batch, running native segment for import/ClipToWindow/Area/TRTEngineOp_0**

This is it. It runs but alsways took 1.8 seconds/image (size 1024x) which is terrible ! The message above keep poping up that batch_size is 153216 while I submit only ONE SINGLE image !!!

I do not think that TensorRT is for PRODUCTION level just yet !

KelvinCPChiu, 2020-01-23:
I could not even run the inference putting the network in the Jetson Xavier. Could you show the code to let me know how to make the inference run ? Thx.

Rub21, 2020-12-16:
I am facinig  same issue. I have exported my module with tha flag: `input_type=encoded_image_string_tensor` and the output files were exported to a docker container  and running it as: `docker run -p 8501:8501 -it aiaia_human_activities:v1.1-cpu` 

The inference takes ~100 seconds to process 2 images, I have tried  soem altternatives, such us rebuilding the  tf serving with the flag:  `TF_SERVING_BUILD_OPTIONS="--copt=-mavx --copt=-mavx2 --copt=-mfma --copt=-msse4.1 --copt=-msse4.2"` and adding warm up tf_serving_warmup_requests file for starting the module, but still slow.

[image]