NVIDIA-AI-IOT/tf_trt_models #83: jetson nano inference

8787878877, 2021-05-13:
I'm sorry, I want to ask if I made inference on jetson nano, the result I ran out was as shown in the picture below, is there any way to solve it[
[image]
](url)

BhagyashreeKoragi, 2021-05-13:
My first question is if we follow the steps, can we actually run these models on Jetson Nano with jetpack 4.4?

8787878877, 2021-05-13:
Hello, my JETPACK is version 4.2, it's adifferent version, is there a way to run with this version, thank you

BhagyashreeKoragi, 2021-05-13:
I'm following the same steps that have been mentioned for jetson tx2.
And trying to run the inceptionv2 model on jetson nano.

In case this might help you. [https://spyjetson.blogspot.com/2019/09/jetsonnano-installing-tensorflow-114.html](url). This is the link to find Tensorflow wheel for jetpack 4.x versions.

8787878877, 2021-05-13:
When I installed TENSORFLOW-GPU before, Iinstalled it on this web page

 
 

In case this might help you. https://spyjetson.blogspot.com/2019/09/jetsonnano-installing-tensorflow-114.html. This is the link to find Tensorflow wheel for jetpack 4.x versions.

8787878877, 2021-05-13:
Hello,my tensorflow-gpu uses version 1.14, thank you

BhagyashreeKoragi, 2021-05-13:
@8787878877  I don't understand your problem clearly. You have to provide a bit more information.

8787878877, 2021-05-13:
_Hello, my jetpack is 4.21, tensorflown-gpuis 1.14, cuda is 10.0, tensorRT is version 5.1.1, the model I use is
Ssd_Mobilenet_V2, I havetrained myself, then the result after inference is the same as the followingfigure, then what method can I use to solve it, thank you

8787878877, 2021-05-13:
Hello, I am my program

8787878877, 2021-05-17:
Sorry, I want to ask, I changed my trainingmodel to ssd_mobilenet_v1_coco. It can run normally when using the sampleprogram, but when I use the model I trained, it can’t display the frame andaccuracy. Is there any What solution